mcu-debug/memview #1: STM32 memory locations off by 10 bytes

meltdown03, 2022-10-07:
I confirmed the memory locations in the STM32411CE Reference Manual as well, but the memories displayed by "memview" are all off by 10 bytes as you can see in this screenshot. "memview" shows the ARR register in 0x4001001c instead of 0x4001002c.

[image]

rubenstar, 2022-10-11:
I can confirm this offset of 0x10 on my setup. Writing location 0x0e001830 to all F's will read correct for CortexDebug v1.6.4, but fail vor v1.6.6.

Old MemoryView on v1.6.4
[image]

New MemoryView on v1.6.6
[image]

haneefdm, 2022-10-11:
Hmmm. Looking at it now...Thanks for reporting

haneefdm, 2022-10-11:
Can you try the latest version 0.0.14, please?

Should be in the marketplace now.

rubenstar, 2022-10-12:
Fixed it for me! Thanks a lot!

rubenstar, 2022-10-12:
Fixed it for me, thanks!

closed 2022-10-12 by haneefdm

meltdown03, 2022-10-13:
Fixed here as well