mcu-debug/memview #2: show wrong data

hongshui3000, 2022-10-13:
The data at 0x10000000 address is 0x96f3b83d, but the memview shows all 0s
[image]
[image]
log：
[logs.txt](https://github.com/mcu-debug/memview/files/9774971/logs.txt)

haneefdm, 2022-10-13:
Is this with the latest version of this extension? Please check to make sure you are using the latest 0.0.14+

I am thinking this is the same as #1

hongshui3000, 2022-10-14:
> Is this with the latest version of this extension? Please check to make sure you are using the latest 0.0.14+
> 
> I am thinking this is the same as #1

[image]

haneefdm, 2022-10-14:
That is not my extension!!?!? Ours extension id is mcu-debug.memory-view (MemoryView)

Please make sure you uninstall the OBSOLETE MemoryView -- haneefdm.memview

hongshui3000, 2022-10-14:
Updating to the latest extension seems to work

closed 2022-10-14 by hongshui3000