mcu-debug/memview #6: Using C expressions to specify the mem doesn't work

Norm369, 2022-10-18:
In the mem viewer in 1.6.4 you could just enter the name of an array (and its size) and it would be displayed. That doesn't seem to work in this extension. 
[image]
[image]

One image shows cortex-debug 1.6.4, with the array g_iAbridgedADCVals displayed. The other shows the memview error message when I attempt to display the same thing. The array is a global.

haneefdm, 2022-10-18:
In Cortex-Debug, memory viewer was a one step query with GDB. Now, it is a two step process.

1. Convert `expression` to an address (as gdb to evaluate the expression and return the result)
2. Use that address to fetch memory.

This is done so that we are compatible with all debuggers. This memory viewer is designed to work with all debuggers and uses the [Debug Adapter Protocol (DAP)](https://microsoft.github.io/debug-adapter-protocol/overview). In Cortex-Debug, we had a DAP but we implemented a custom memory fetcher because at the time the DAP didn't have a memory fetcher. Now it does and all debuggers are using it.

Try `&g_iAbridgedADCVals` for the expression

haneefdm, 2022-10-19:
I am closing this. Make sure any expression you enter will result in an address. Arrays (and structs) need an `&` or gdb will not print an address for an array.

Reply if it does not work.

closed 2022-10-19 by haneefdm