mcu-debug/memview #7: Not vertical layout friendly

diggit, 2022-10-27:
Hi,
memview does not work well with high and narrow panels.

[image]

It automatically hides addresses and leaves ASCII decoding enabled. Top bar is not properly aligned with contents below.
I'd rather see addresses than ASCII decoding. Preferably configurable behavior.

Also, memview does not readjust its' width, when panel changes dimensions. Manual memview refresh has to be done.

It's great to see active development of memory view :)

haneefdm, 2022-10-27:
Hmmm. Not even sure how this is possible. By any chance, the addresses showed up on far right?

>  Also, memview does not readjust its' width, when panel changes dimensions. Manual memview refresh has to be done.

Also, noted.

haneefdm, 2022-10-29:
I cannot duplicate this issue. I can make it pretty narrow and it still works. Do you have steps to reproduce? What platform is it (although it is all the same HTML and the same browser). I also want to make sure you are using the right extension and are using the latest version. This extension's unique ID is `mcu-debug.memory-view`. I have an obsolete one but you will see that marked as such in the README for the extension.

[image]

diggit, 2022-11-01:
Hi,
I have installed: mcu-debug.memory-view `v0.0.14` (Preview)

here are some details about behavior:

Moved panel to right, made it wide enough and refreshed memory view
[image]

Shrunk a little bit. Individual addresses became scrollable.
[image]

And even more...
[image]

And then they completely disappear.
[image]

Memview becomes horizontally scrollable to fit contents and ascii decode, but addresses are nowhere to be seen.

I guess, something is missing "minimum width" property or so.

haneefdm, 2022-11-12:
I pushed an update to the marketplace just now 0.0.16. See Changelog. The header does not horizontally scroll but the content does. We don't know what is wrong or how to fix it and seeking help from the community.

But, the vertical bar works pretty well in our testing.

Best we can do for now.

haneefdm, 2022-11-12:
I am not gong to close this but please give us feedback if you have the time

hongshui3000, 2022-11-16:
I am showing incomplete data here  0.0.16
[image]

haneefdm, 2022-11-16:
@hongshui3000 Did you change the editor font-size in VSCode by any chance? Was this the same with the previous version? It appears that the line-height is not enough. I am not sure how to duplicate this.

Btw, this is a separate issue from the original issue of very narrow windows.

hongshui3000, 2022-11-16:
font-size
[image]

haneefdm, 2022-11-18:
line spacing issue is fixed in latest release in the marketplace.

haneefdm, 2022-11-22:
I am closing this for now. The horizontal scrolling could be a separate issue and not as pressing. The original issue has long been solved and I don't want people to use this issue to add other stuff.

closed 2022-11-22 by haneefdm