mcu-debug/memview #9: Data regions are not refreshed synchronously

hongshui3000, 2022-11-16:
[image]
Data regions are not refreshed synchronously

[image]
When I compile the address and display name

hongshui3000, 2022-11-16:
、

> You changed both address and display name? 
yes，The data is not refreshed after the change

> Does a manual refresh (button) work?
yes

haneefdm, 2022-11-16:
Should be fixed in the next update. The view contents will be updated only if
* The debugger session is in a 'stopped' state
* Or else it will be updated the next time the debugger enters the 'stopped' state -- an info bubble will pop up

haneefdm, 2022-11-18:
Just released the update. If you change the expression, the window will update if the program is in the 'stopped' state.

You should also see the memory window update itself with proper line spacing as soon (tiny delay) as you change the font size.

closed 2022-11-18 by haneefdm