mcu-debug/memview #10: Misaligned rows when saving memory to file

danielaird27, 2023-02-01:
As shown in the image below, when a user saves the memory as a file (txt file) using the "Save to file..." button in the "Memory" tab, the address column is misaligned with the data.

For example, the row with the address ending in 0x110 in the "Memory" tab in VSCode shows byte 0 as 0x43 but the saved file shows that same byte at the address ending in 0x100 instead.

[image]

haneefdm, 2023-02-01:
Hmmm Weird. The problem started in the first two rows. The first address is duplicated in the text output.

Were these shots taken at different times? I see another problem beyond misalignment. The misaligned row does not have same data for columns 0x04

danielaird27, 2023-02-01:
Yes, they were taken at slightly different times, I just use them to showcase the misaligned rows. The data for columns 0x04 isn't an issue.

haneefdm, 2023-02-02:
Thanks for confirming. I was super scared when I saw that.

haneefdm, 2023-02-02:
Should be fixed. I pushed the update to the marketplace. You should be getting it in a few hours...

Please close this issue after confirming the fid.

haneefdm, 2023-02-02:
Look for version V0.0.19

closed 2023-03-10 by haneefdm

danielaird27, 2023-03-13:
@haneefdm Sorry for the late reply, the issue is fixed, thank you!