mcu-debug/memview #11: Non-global variables can't be captured.

Eplankton, 2023-02-04:
I read the manual and noticed that global variables are recommended here instead of local variables, and I just have a test on it, got stuck at certain point there, whether in C or C++, can we know exactly why this happens ? Why the viewer works fine on global but not on local?

haneefdm, 2023-02-04:
Well, local variables go in and out of scope. It may work but things can fail when you continue and switch scopes. Local variables by definition are transient

If GDB can't evaluate your expression and resolve to an address, we can't read memory from that address.

Maybe, I do not understand your question

haneefdm, 2023-02-04:
As you know, you can always use an address to get memory contents from. Closing this issue unless there are more details as why this is an issue or som request for an enhancement

Eplankton, 2023-02-05:
> 

I believe that you have misunderstood my question, surely a local variable can't be detected outside its lifetime scope, what i want to say is, for example, here's the source code:
```
void foo()
{
    ...
    int a[] = {1, 2, 3, 4, 5}; // declare an array
    a[0]++; // change the value, expect the map viewer to show its changes
    ...
}
```
and I try to give the map viewer `&a` when the debugger stops  in the context behind the viewer fails to read the address of array `a` and show the change of value.

Eplankton, 2023-02-05:
> 如您所知，您始终可以使用地址从中获取内存内容。 关闭此问题，除非有更多详细信息说明为什么这是一个问题或请求增强

In practice, as showed in the picture, though the GDB can correctly parse the address of `a`, what i got is just endless `nul`
[image]

haneefdm, 2023-02-05:
It was likely optimized out. What do you see gdb return. Depending on the debugger, try to enable their debug mechanism to see gdb traces. Also, what does the debugger show for 'a'?

Eplankton, 2023-02-06:
> 它可能被优化了。你看到 gdb 返回什么。根据调试器，尝试启用其调试机制以查看 gdb 跟踪。另外，调试器为“a”显示什么？

In comparsion, if i just move array outside and turn it into a global variable, then the problem fixed:
[image]

haneefdm, 2023-02-06:
Ahh, is this variable on the top of the frame? If not, I can see a problem. There is no way to know what the current thread/frame is. But since your print is working in gdb? First two things:

1. Can we make sure `&a` resolves to the same address via gdb?
2. It appears you are using cortex-debug. Can you set the debug output to "raw" and capture everything after your single step? Clear Console, single step, copy the new output into a file, and attach it here. You will see all the debug transactions and the requests we made. You may see a lot of output unrelated to the memory view, but that is okay.

Eplankton, 2023-02-06:
> 
For Question 1, I can be sure that these two address are exactly same.
For Question 2：

```
Cortex-Debug: VSCode debugger extension version 1.6.10 git(0458417). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from C:/Users/Wike Chao/.eide/tools/gcc_arm/bin/arm-none-eabi-objdump.exe --syms -C -h -w D:/code/EIDE/new/build/Debug/new.elf
Reading symbols from c:/users/wike chao/.eide/tools/gcc_arm/bin/arm-none-eabi-nm.exe --defined-only -S -l -C -p D:/code/EIDE/new/build/Debug/new.elf
Launching GDB: "C:\\Users\\Wike Chao\\.eide\\tools\\gcc_arm\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2
1-gdb-version
Launching gdb-server: "C:\\Program Files (x86)\\OpenOCD\\bin\\openocd.exe" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "D:\\code\\EIDE\\new" -f "c:/Users/Wike Chao/.vscode/extensions/marus25.cortex-debug-1.6.10/support/openocd-helpers.tcl" -f interface/stlink.cfg -f target/stm32f4x.cfg
    Please check TERMINAL tab (gdb-server) for output from C:\Program Files (x86)\OpenOCD\bin\openocd.exe
Finished reading symbols from objdump: Time: 72 ms
Finished reading symbols from nm: Time: 68 ms
C:\Users\Wike Chao\.eide\tools\gcc_arm\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
-> =thread-group-added,id="i1"
-> ~"GNU gdb (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.1.90.20201028-git\n"
-> ~"Copyright (C) 2020 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=i686-w64-mingw32 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source c:/Users/Wike Chao/.vscode/extensions/marus25.cortex-debug-1.6.10/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source c:/Users/Wike Chao/.vscode/extensions/marus25.cortex-debug-1.6.10/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-file-exec-and-symbols "D:/code/EIDE/new/build/Debug/new.elf"
-> 8^done
9-target-select extended-remote localhost:50000
-> =thread-group-started,id="i1",pid="42000"
-> =thread-created,id="1",group-id="i1"
-> ~"loop () at ./src/main.cpp:124\n"
loop () at ./src/main.cpp:124
-> ~"124\t\t\ta[0]++;\n"
124			a[0]++;
-> *stopped,frame={addr="0x08004c1a",func="loop",args=[],file="./src/main.cpp",fullname="d:\\code\\EIDE\\new\\src\\main.cpp",line="124",arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
Program stopped, probably due to a reset and/or halt issued by debugger
-> 9^connected
10-interpreter-exec console "monitor reset halt"
-> @"Unable to match requested speed 2000 kHz, using 1800 kHz\n"
Unable to match requested speed 2000 kHz, using 1800 kHz
-> @"Unable to match requested speed 2000 kHz, using 1800 kHz\n"
Unable to match requested speed 2000 kHz, using 1800 kHz
-> @"target halted due to debug-request, current mode: Thread \n"
target halted due to debug-request, current mode: Thread
-> @"xPSR: 0x01000000 pc: 0x0800499c msp: 0x20030000\n"
xPSR: 0x01000000 pc: 0x0800499c msp: 0x20030000
-> 10^done
11-target-download
-> 11+download,{section=".isr_vector",section-size="428",total-size="184406"}
-> 11+download,{section=".isr_vector",section-sent="428",section-size="428",total-sent="428",total-size="184406"}
[... 137 lines cut ...]
-> 53^error,msg="Variable object not found"
54-var-create --thread 1 --frame 0 undefined_57671eae1c04a1651b3c8069e9e72d2475467713ecb6d22efc0b44ea6aa0b4d2 * "_tx_thread_created_ptr"
-> 54^error,msg="-var-create: unable to create variable object"
undefined -var-create: unable to create variable object (from var-create --thread 1 --frame 0 undefined_57671eae1c04a1651b3c8069e9e72d2475467713ecb6d22efc0b44ea6aa0b4d2 * "_tx_thread_created_ptr")55-var-update --thread 1 --frame 0 --all-values undefined_c94481adaea4ef65c72e1e8d1bb7dfdd3a7c97dfad36adc0e694b8026d37a09f
-> 55^error,msg="Variable object not found"
56-var-create --thread 1 --frame 0 undefined_c94481adaea4ef65c72e1e8d1bb7dfdd3a7c97dfad36adc0e694b8026d37a09f * "pxReadyTasksLists"
-> 56^error,msg="-var-create: unable to create variable object"
undefined -var-create: unable to create variable object (from var-create --thread 1 --frame 0 undefined_c94481adaea4ef65c72e1e8d1bb7dfdd3a7c97dfad36adc0e694b8026d37a09f * "pxReadyTasksLists")57-var-update --thread 1 --frame 0 --all-values undefined_5e0c067643e216edacb586fb651891530718e7c644a121886d01aebb41c29362
-> 57^error,msg="Variable object not found"
[... cut ...]
undefined -var-create: unable to create variable object (from var-create --thread 1 --frame 0 undefined_6ef43a260eab87240c7916427d4fbe5dbe871dc760b9ad8899bca3567c3bc096 * "_track_list_k_mutex")61-stack-select-frame --thread 1 0
-> 61^done
62-stack-list-variables --thread 1 --frame 0 --simple-values
-> 62^done,variables=[{name="x",type="nuts::UART"},{name="a",type="int [5]"}]
63-var-update --thread 1 --frame 0 --all-values var_x_256
-> 63^done,changelist=[]
64-var-update --thread 1 --frame 0 --all-values var_a_256
-> 64^done,changelist=[]

```

haneefdm, 2023-02-06:
```
66-var-create --thread 1 --frame 0 hover_b5d61e064f9830cea01eed28b25700200331bdf4791f8d75bcd25ded432aebd4 * "&a"
-> 66^done,name="hover_b5d61e064f9830cea01eed28b25700200331bdf4791f8d75bcd25ded432aebd4",numchild="1",value="0x2002ffd0",type="int (*)[5]",thread-id="1",has_more="0"
```

This says value of `&a` is `0x2002ffd0` -- and the current frame was used. The current frame definition can be anything and in VSCode, it is undefined and may change based on your last action in the debug panel. This has been a limitation of VSCode for a long time. There is one other option to try as an experiment, we need to finish our investigation.

```
69-data-read-memory-bytes "0x2002ffd0" 512
-> 69^done,memory=[{begin="0x2002ffd0",offset="0x00000000",end="0x200301d0",contents="0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"}]
70-data-read-memory-bytes "0x200301d0" 512
-> 70^done,memory=[{begin="0x200301d0",offset="0x00000000",end="0x200303d0",contents="0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"}]
```

Above says the memory at address `0x2002ffd0` is all zeros. So, the address must be wrong or gdb is lying to us with either the address or the content

> For Question 1, I can be sure that these two address are exactly same.

Please humor me and do a `print &a` and report back. You maybe sure, but we don't know what gdb thinks.

haneefdm, 2023-02-06:
Do you have another variable `a` in your program -- perhaps in a global scope?

Eplankton, 2023-02-06:
I even try to just printf the address through UART, and got same:
```
==========================================================================
port: COM4, baud rate: 9600, bit width: 8, parity: None, stop bits: One
==========================================================================

2002ffd0
```
Here's gdb output, I tried many different ways to ensure that `0x2002ffd0` is valid:
```
print a
$5 = {1, 2, 3, 4, 5}
{"output":"","token":61,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}

print &a
$6 = (int (*)[5]) 0x2002ffd0
{"output":"","token":66,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}

print *(0x2002ffd0)
$7 = 1
{"output":"","token":71,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}

print *(0x2002ffd0)@5
$8 = {3, 2, 3, 4, 5}
```
[image]

Eplankton, 2023-02-06:
> Do you have another variable `a` in your program -- perhaps in a global scope?

Definitely not, I have checked the list of global variable.

haneefdm, 2023-02-06:
But as you can see, gdb is telling us that it is all zeros at that address, and we didn't make that up. I cannot explain that.

Another possibility is that your variable `a` is all in registers and nothing was ever written back to memory/stack. I see that you did not declare it as volatile in some cases and then, yes that can happen. Given the code, I see there is no reason for GCC to generate instructions to update the memory/stack. You can look at the disassembly to ensure.

Another question. Can you see in the gdb-server TERMINAL tab, any memory requests (they come from gdb). If not, someone is caching previous results. You can use `serverArgs` to specify additional command-line args to openocd. I think the option to use is `-d3` or higher. Our memory read request path is -> vscode -> cortex-debug -> gdb -> openocd and then all the way back.

Eplankton, 2023-02-06:
> 但正如您所见，gdb 告诉我们该地址全为零，而我们并没有编造它。 我无法解释。
> 
> 另一种可能性是你的变量 `a`都在寄存器中，没有任何内容被写回内存/堆栈。 我看到您在某些情况下没有将其声明为易变的，然后，是的，这可能会发生。 根据代码，我发现 GCC 没有理由生成更新内存/堆栈的指令。 您可以查看反汇编以确保。
> 
> 另一个问题。 您可以在 gdb-server TERMINAL 选项卡中看到任何内存请求（它们来自 gdb）。 如果不是，则有人正在缓存以前的结果。 您可以使用 `serverArgs`为 openocd 指定额外的命令行参数。 我认为使用的选项是 `-d3`或更高。 我们的内存读取请求路径是 -> vscode -> cortex-debug -> gdb -> openocd 然后一直往回走。
> 
> [https://openocd.org/doc/html/Running.html#Running ](https://openocd.org/doc/html/Running.html#Running)

With disasmble code:
```
	volatile int a[] {1, 2, 3, 4, 5};
  12:	4b09      	ldr	r3, [pc, #36]	; (38 <_Z4loopv+0x38>)
  14:	463c      	mov	r4, r7
  16:	461d      	mov	r5, r3
  18:	cd0f      	ldmia	r5!, {r0, r1, r2, r3}
  1a:	c40f      	stmia	r4!, {r0, r1, r2, r3}
  1c:	682b      	ldr	r3, [r5, #0]
  1e:	6023      	str	r3, [r4, #0]
```

haneefdm, 2023-02-06:
This disassembly is for when you declared it volatile. But in your code I saw above, where bug was visible, it was declared as normal variable (not volatile). See below

[image]

If you want to snoop on memory like we are trying to do, It has a very low chances for local variables unless they are declared as volatile

haneefdm, 2023-02-06:
This disassembly is for when you declared it volatile. But in your code I saw above, where bug was visible, it was declared as normal variable (not volatile). See below

[image]

If you want to snoop on memory like we are trying to do, It has a very low chance for local variables to work unless they are declared as volatile. Even then gcc is sometimes too smart.

Eplankton, 2023-02-06:
> 此反汇编适用于您将其声明为 volatile 的情况。 但是在我上面看到的你的代码中，错误是可见的，它被声明为普通变量（不是易变的）。 见下文
> 
> [image]
> 
> 如果你想像我们正在尝试的那样窥探内存，它对局部变量的机会非常小，除非它们被声明为 volatile

But even if i put a `volatile` there, nothing changes.......

haneefdm, 2023-02-06:
All I can say is with volatile, you have some chance of it working and without it an extremely low chance of it working.

I don't think we have an issue here that I can act on. We already knew this issue and documented it. The issue is with how the compiler generates code and I see no way of working around that. We are also not wrong.

closed 2023-02-06 by haneefdm

Eplankton, 2023-02-06:
> 

Alright...