mcu-debug/memview #12: unbound breakpoints when debuging this extension

Joffygo, 2023-02-14:
Hi, I try to figure out how this extension works. So I git clone the code->npm install->npm run compile and start debug.
however, the breakpoints set in the srcFiles turns into unbound when debugging.
[image]
I've google this unbound breakpoint issue,but the answers in stackoverflow don't suit my suitation.
Thanks!

haneefdm, 2023-02-14:
Never seen that message. Let us know how to fix it if it is a problem with our JSON (launch, task, package, etc.) files. I don't think it has to do with our extension, we didn't do anything special.

And no idea what that file is from. There are two kinds of files, one that runs in the main VSCode context and one that runs in the browser context for the actual view. The one in the browser though may periodically jump to the translated JS rather than the TS, but it still debuggable.

Joffygo, 2023-02-14:
I don't think it is a problem with json files,either. I just don't quite sure whether there are extra jobs need to do before debugging(except npm run compile).
This file I list before is memview-doc.ts. Actually I've set breakpoints in extension.ts, and it doesn't work either.

haneefdm, 2023-02-14:
Only thing I do is run the task `npm watch` withing VSCode so it keeps rebuilding while I edit/edbug.

haneefdm, 2023-02-14:
I do that with all my projects. Have the watch running in the background.

Joffygo, 2023-02-14:
> Only thing I do is run the task `npm watch` withing VSCode so it keeps rebuilding while I edit/edbug.

Wow!  I don't know why. But this works.
[image]
maybe it's about esbuild.
whatever. it works now.
Thank you!

Joffygo, 2023-02-14:
it seems it's about compile command:
change it to tsc -p ./ and it works too.
[image]

haneefdm, 2023-02-14:
No clue why the compile does not work. Yes, we use esbuild and technically it is the same command except without a watch. I can't change the compile command because it is used during extension publishing and I would rather use the same method for both compile and watch. Could it be that the `-p ./` is missing on `tsc`

Curious, about what you are doing with the extension.

Joffygo, 2023-02-15:
[image]
vscode need .js.map file to bind breakpoints. And npm run watch will generate this file. However ,if you delete this file,and run npm run compile, .js.map file doesn't been generated. 

tsc -p ./ actually didn't work either cause it generate .js.map file in Out directory instead of dist directory.
last time I consider tsc -p ./ works is because that I didn't delete .js.map file which is generated by npm run watch.

haneefdm, 2023-02-15:
Okay, I found the reason for this. The idea is that for production, the map file (which can be large) are not to be bundled for release.

https://github.com/mcu-debug/memview/blob/3ff9a5789edc3b799ab5d04830c943d590207def/.esbuild.config.js#L38

If you are in development mode you should be running `npm watch`. Also, not sure if there is a way `launch.json` can enforce that using `preLaunchTask`

haneefdm, 2023-02-15:
I found a workaround that also helps me because it has been a manual step for me whenever I open a folder/workspace. Now, it should solve both your issue and will help me out as well. In `tasks.json` you can have the `watch` task start on folder open...

```JSON
            "runOptions": {
                "runOn": "folderOpen",
	       "instanceLimit": 1
            }
```

I will update this repo when I get a chance. I am closing this for now.

closed 2023-02-15 by haneefdm