mcu-debug/memview #13: Add a button to remove memory view from drop down list

eZioPan, 2023-04-22:
**Is your feature request related to a problem? Please describe.**

I want to remove some memory view from drop down list, but currently there is only add and modify button.

**Describe the solution you'd like**

maybe add a remove button for deleting current memory view? 

**Describe alternatives you've considered**

maybe add button at Edit popup window to delete the memory view?

haneefdm, 2023-04-22:
Did you see the button on the top right of the Window? This is where most close buttons are in VSCode Panels.

eZioPan, 2023-04-22:
oh, i just see that button, thanks!  
But i still think close button is not that obvious, and might be too far away from main function zone. I feel it might be a little better to move the button closer to main function zone.

[image]

haneefdm, 2023-04-22:
I partially agree. But, we are following the VSCode design language. If we add it to the main toolbar, then we have to create/add a prompt because you can accidentally hit it.

I will see if VSCode has a more prominent Close button. Even these, we don't make our own.

closed 2023-04-22 by haneefdm