mcu-debug/memview #15: Work with lldb

ldelossa, 2023-06-21:
Hello,

Does this plugin work with lldb?

haneefdm, 2023-06-22:
I don't see why not. We are actual debugger (lldb, gdb, vsdbg, cspy, etc) agnostic. So, it depends on which extension/debugger you are using. See the top of our readme.

https://github.com/mcu-debug/memview/blob/master/README.md

Your extension has to support the proper APIs to read memory. That is all. If your debugger extension is not in that list, we can easily add that

ldelossa, 2023-06-22:
Thanks.

I actually use this: https://marketplace.visualstudio.com/items?itemName=vadimcn.vscode-lldb since the output is a lot nicer. 

I dont see it as supported, or does this plugin detect that the debugger has started

haneefdm, 2023-06-22:
Yes, since there are a TON of debuggers in VSCode, we don't interfere with most. Only a select few we know of and tested with.

I will enhance the extension to add `your` debugger type(s). Perhaps over the weekend.  I need you to help test it though.

ldelossa, 2023-06-22:
Sounds great :). 

Just ping me when youd like me to test.

haneefdm, 2023-06-24:
Just released an update. May take a few hours for MS to push the update.

See ChangeLog to add your own debuggers.

ldelossa, 2023-06-26:
Thanks. It works with 'lldb'. 

One other unrelated question, can I have it just print '0' instead of the word 'null' ?

closed 2023-06-26 by haneefdm

ldelossa, 2023-06-26:
I dont think I follow. NULL *is* zero in memory right? And memory view doesnt care about data type, so why replace 0 with nul?

ldelossa, 2023-06-26:
Ah I'm an idiot, just realized I was looking at the "decoded bytes" section for ASCII. Ignore me.