mcu-debug/memview #16: Refresh not done when a breakpoint is done

xamix, 2023-07-13:
**Describe the bug**
Refresh is not correctly done when hit a breakpoint (but it work when click on refresh button and when click on pause button)

**To Reproduce**
Steps to reproduce the behavior:
1. Launch a debug session and open a memory view on an area you know data change at every breakpoint
2. Hit multiple time on the breakpoint and check if the memory value change
3. Memory is not updated, until you click on reload or press pause on debugger

See the following GIF

**Expected behavior**
When the breakpoint hit, normally the view should refresh

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: Windows
 - Version of Memory View: mcu-debug v0.0.22
 - Debugger Type: cppdbg

haneefdm, 2023-07-14:
For `cppdbg`, can you enable the various kinds of logging available? I don't use `cppdbg` so, I can't tell you. There should be an `launch.json` option to enable debug transactions and yet another to see VSCode transactions. See...

https://code.visualstudio.com/docs/cpp/launch-json-reference#_logging

After you encounter a breakpoint, we should be generating requests for memory reads. Is that happening? If so, what does gdb/lldb/vddbg (not sure which backend you are using) report/return?

If we are making the request and we are getting wrong results, then there is nothing we can do.

hongshui3000, 2023-07-14:
I think it is reasonable to automatically refresh the current mem window when a breakpoint is encountered, but I don't know if the extension can obtain breakpoint information

haneefdm, 2023-07-14:
Of course, we always do an auto refresh. Why it is not working with this debugger is the question.

We don't need breakpoint info for that. As you can see we keep track of the current status of the debug session.

xamix, 2023-07-15:
I use MSPDepug backend to test, and it print debugger access so for what I can say, the MemoryView do not send any request to the GDB server.

When I run and hit a breakpoint here is the LOG:

[image]

As you can see no update of the memory is done, only reading of local variable from `VARIABLES`  panel seem to be done automatically by the debugger.

If I go to the memory view  and click on the UPDATE button, then I can see the reading occuring:

[image]

haneefdm, 2023-07-15:
Thanks for the update. We were able to reproduce the problem. We released an experimental fix to the marketplace which should be available shortly.

This would not affect embedded debuggers -- it only affected very fast computers running native code. See also the ChangeLog

Please give us feedback and close this issue if it is working okay for you.

xamix, 2023-07-16:
Hi, 

It seem to work now, so issue is FIXED for me.

Thank you for your reactivity.

closed 2023-07-16 by xamix