mcu-debug/memview #18: No data in around SP (Stack Pointer) on Pi Pico unless I show the previous 425 or more bytes

meltdown03, 2023-07-16:
Using the latest version of the extension: v0.0.23 on multiple Pico examples

To reproduce: I step through the code in disassembly and watch for push instructions as I watch the SP memory locations. I add the location `$sp` in the MemoryView dialog box. 
Result:
It is always all zeros. I know the stack is backwards so I use `$sp-32` or something similar depending on how far back I think I should go. All zeros again. I found the magic number is `$sp-425`, then the memory is shown including all the way up to the SP address. `$sp-424` is all zeros.
I have 2 screenshots of this, one using 32 bytes under SP and the other using 512 bytes. I highlighted the same address in memory and you can see that they are different.
[image]
[image]

haneefdm, 2023-07-16:
Does this happen with v0.0.22?

Did you enable logging/debug on your debugger to show gdb traces? See what gdb transactions look like?

meltdown03, 2023-07-16:
Yes it was happening with the last version. If I examine the memory with
gdb it shows the correct info.
Do you mean the logLevel in the launch.json file? I can do that.

haneefdm, 2023-07-16:
If you are using Cortex-Debug, it would be `showDevDebugOutput` like it recommends in the Debug Console. You need to attach what you see in the debug console. All we do is make a request to the debugger. It is up to the debugger to fetch the memory (in this case gdb) and send it to us.

Note that $sp does not have a specific meaning. Because it is frame dependent. If you switch stack frames, the $sp/$pc changes. We always try to use the top frame as we have no idea what frame you have selected in the Stack Trace window.

haneefdm, 2023-07-16:
Here is how we debug things and you HAVE to use the facilities your debugger provides. I have no idea which debugger you are using but here is an example with a `cppdbg`.

https://github.com/mcu-debug/memview/issues/16

You have to do some investigation as the submitter of the above issue did. It is also hard when you provide so little info. You did not use our issue submission template. Please do so next time.

meltdown03, 2023-07-16:
Sorry about that, It is on Linux using Cortex-Debug, launch.json:
``` json
{
  "version": "0.2.0",
  "configurations": [
      {
          "name": "Pico Debug",
          "cwd": "${workspaceRoot}",
          "executable": "${command:cmake.launchTargetPath}",
          "request": "launch",
          "type": "cortex-debug",
          "servertype": "openocd",
          // This may need to be arm-none-eabi-gdb depending on your system
          "gdbPath" : "arm-none-eabi-gdb",
          "device": "RP2040",
          "showDevDebugOutput": "both",
          "configFiles": [
              "interface/cmsis-dap.cfg",
              "target/rp2040.cfg"
          ],
          "svdFile": "/home/neal/pico/pico-sdk/src/rp2040/hardware_regs/rp2040.svd",
          "runToEntryPoint": "main",
          // Work around for stopping at main on restart
          "postRestartCommands": [
              "break main",
              "continue"
          ],
          "openOCDLaunchCommands": ["adapter speed 5000"],
          "searchDir": ["/usr/share/openocd/scripts/"],
      }
  ]
}
```

 here is the area of the log output, did you want to see the entire log?
```
167^done,memory=[{begin="0x20041d00",offset="0x00000000",end="0x20041f00",contents="8e55ea935bc5112d09648cd91e0000000d38a0f70001000000000000a81e04200000000004000000901d0420000000000000000035210010901d0420040000000000000000000000830e8ffd000000003200000031000000af1f0010bb1f0010320000002d310010901d042004000000000000000a000000000000000000000000000000fe0d940b1e000000000100003032303200000000d95c887c577973a2521cb203b341001000000000fa4b000a010000001d44001031000000f81d0420320000008740001000000000d0020020f41d04209c1e042000000000f740001032000000154200103200000059430010e06a1ce900000000416c61726d204669726564204174205765646e6573646179203133204a616e756172792031313a32313a303020323032300a0000ffffffffffffffff6c1e0420a51f00100d0000000b0000005f310010881e04201b5f8719b9fa78b66c1e042055f8a7873c530010000000002f160010dc520010105300100d0000003c530010a81f0420e81f042000000018714300105080a81e9c1e0420ffffffff07030010e84b0010a81e0420a81f04200b0300100000000008000240da0000005a000000cc1e042079240000000300000100000004000000001f0420180300209c020020010000008fcb2d009c02002011180010180300209c0200209c020020550a0010401f0420850c0010"}]
GDB -> App: {"output":"","token":167,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["memory",[[["begin","0x20041d00"],["offset","0x00000000"],["end","0x20041f00"],["contents","8e55ea935bc5112d09648cd91e0000000d38a0f70001000000000000a81e04200000000004000000901d0420000000000000000035210010901d0420040000000000000000000000830e8ffd000000003200000031000000af1f0010bb1f0010320000002d310010901d042004000000000000000a000000000000000000000000000000fe0d940b1e000000000100003032303200000000d95c887c577973a2521cb203b341001000000000fa4b000a010000001d44001031000000f81d0420320000008740001000000000d0020020f41d04209c1e042000000000f740001032000000154200103200000059430010e06a1ce900000000416c61726d204669726564204174205765646e6573646179203133204a616e756172792031313a32313a303020323032300a0000ffffffffffffffff6c1e0420a51f00100d0000000b0000005f310010881e04201b5f8719b9fa78b66c1e042055f8a7873c530010000000002f160010dc520010105300100d0000003c530010a81f0420e81f042000000018714300105080a81e9c1e0420ffffffff07030010e84b0010a81e0420a81f04200b0300100000000008000240da0000005a000000cc1e042079240000000300000100000004000000001f0420180300209c020020010000008fcb2d009c02002011180010180300209c0200209c020020550a0010401f0420850c0010"]]]]]}}
168-data-read-memory-bytes "0x20041f00" 512
-> 168^done,memory=[{begin="0x20041f00",offset="0x00000000",end="0x20042100",contents="0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"}]
GDB -> App: {"output":"","token":168,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["memory",[[["begin","0x20041f00"],["offset","0x00000000"],["end","0x20042100"],["contents","000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...]
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"0x20041ee0:\t0x2000029c\t0x10001811\t0x20000318\t0x2000029c\n"}]}
0x20041ee0:	0x2000029c	0x10001811	0x20000318	0x2000029c
-> ~"0x20041ef0:\t0x2000029c\t0x10000a55\t0x20041f40\t0x10000c85\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"0x20041ef0:\t0x2000029c\t0x10000a55\t0x20041f40\t0x10000c85\n"}]}
0x20041ef0:	0x2000029c	0x10000a55	0x20041f40	0x10000c85
-> 175^done
```

meltdown03, 2023-07-16:
So maybe its the debugger since it's receiving all 0's from and just displaying them like it's told. It's just so odd that if I go back 425 bytes from the stack point address, it works.

meltdown03, 2023-07-16:
I also looked in the Cortex-debug log and matched the logs, first with a working location, then I viewed closed the the SP and got all blank data.

showDevDebugOutput:
```

107-data-read-memory-bytes "0x20041dd0" 512
-> 107^done,memory=[{begin="0x20041dd0",offset="0x00000000",end="0x20041fd0",contents="f41d04209c1e042000000000f740001032000000154200103200000059430010e06a1ce900000000416c61726d204669726564204174205765646e6573646179203133204a616e756172792031313a32313a303020323032300a0000ffffffffffffffff6c1e0420a51f00100d0000000b0000005f310010881e04201b5f8719b9fa78b66c1e042055f8a7873c530010000000002f160010dc520010105300100d0000003c530010a81f0420e81f042000000018714300105080a81e9c1e0420ffffffff07030010e84b0010a81e0420a81f04200b0300100000000008000240da0000005a000000cc1e042079240000000300000100000004000000001f0420180300209c0200200100000092cb2d009c02002011180010180300209c0200209c020020550a0010401f0420850c0010c8cb2d000000000084020020770e001000000000840200200100000001100010410c001000000000000000000000000000000000000000000000000003000000c8cb2d000000000000201866ffffffffffffffffffffffffffffffff880800200c00000000000018000000007713001086cb2d000000000000000018f9ffffff000000000000000000000000280100d0281f04205b12001029490010cd49001000000000000000000100000000000020f40f0420294900106402001064020010011f04200000001800000000fb490010"}]
GDB -> App: {"output":"","token":107,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["memory",[[["begin","0x20041dd0"],["offset","0x00000000"],["end","0x20041fd0"],["contents","f41d04209c1e042000000000f740001032000000154200103200000059430010e06a1ce900000000416c61726d204669726564204174205765646e6573646179203133204a616e756172792031313a32313a303020323032300a0000ffffffffffffffff6c1e0420a51f00100d0000000b0000005f310010881e04201b5f8719b9fa78b66c1e042055f8a7873c530010000000002f160010dc520010105300100d0000003c530010a81f0420e81f042000000018714300105080a81e9c1e0420ffffffff07030010e84b0010a81e0420a81f04200b0300100000000008000240da0000005a000000cc1e042079240000000300000100000004000000001f0420180300209c0200200100000092cb2d009c02002011180010180300209c0200209c020020550a0010401f0420850c0010c8cb2d000000000084020020770e001000000000840200200100000001100010410c001000000000000000000000000000000000000000000000000003000000c8cb2d000000000000201866ffffffffffffffffffffffffffffffff880800200c00000000000018000000007713001086cb2d000000000000000018f9ffffff000000000000000000000000280100d0281f04205b12001029490010cd49001000000000000000000100000000000020f40f0420294900106402001064020010011f04200000001800000000fb490010"]]]]]}}
108-data-read-memory-bytes "0x20041fd0" 512
-> 108^done,memory=[{begin="0x20041fd0",offset="0x00000000",end="0x200421d0",contents="0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"}]
GDB -> App: {"output":"","token":108,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["memory",[[["begin","0x20041fd0"],["offset","0x00000000"],["end","0x200421d0"],["contents","0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"]]]]]}}
```

And Cortex-Debug Tracker activated:
```
<-- {"command":"readMemory","arguments":{"memoryReference":"537140688","count":512},"type":"request","seq":42}
--> {"seq":597,"type":"event","event":"output","body":{"category":"stderr","output":"107-data-read-memory-bytes \"0x20041dd0\" 512\n"}}
<-- {"command":"readMemory","arguments":{"memoryReference":"537141200","count":512},"type":"request","seq":43}
[... cut ...] AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}}
```

meltdown03, 2023-07-16:
I just tried the Legacy memory viewer and it shows the data as expected:

Cortex log
```
<-- {"command":"read-memory","arguments":{"address":"$sp-32","length":32},"type":"request","seq":39}
--> {"seq":352,"type":"event","event":"output","body":{"category":"stderr","output":"77-data-read-memory-bytes \"$sp-32\" 32\n"}}
--> {"seq":353,"type":"event","event":"output","body":{"category":"stderr","output":"-> 77^done,memory=[{begin=\"0x20041fc0\",offset=\"0x00000000\",end=\"0x20041fe0\",contents=\"011f04200000001800000000fb49001064020010354a0010640200108d030010\"}]\n"}}
--> {"seq":354,"type":"response","request_seq":39,"command":"read-memory","success":true,"body":{"startAddress":"0x20041fc0","endAddress":"0x20041fe0","bytes":[1,31,4,32,0,0,0,24,0,0,0,0,251,73,0,16,100,2,0,16,53,74,0,16,100,2,0,16,141,3,0,16]}}
```

showDevDebugOutput
```
77-data-read-memory-bytes "$sp-32" 32
-> 77^done,memory=[{begin="0x20041fc0",offset="0x00000000",end="0x20041fe0",contents="011f04200000001800000000fb49001064020010354a0010640200108d030010"}]
```

meltdown03, 2023-07-16:
I think I found the issue. `Error: Failed to read memory at 0x20042000` is logged in the gdb-server output when I start at the SP but ask to see 512 bytes in lagacy mode. This is like the how the new memory viewer would do the request. I get blank data now. It must be a limitation with OpenOCD so it's not an issue with your extension. Sorry. Maybe you can allow a smaller section of memory to be viewed with an option.

```
<-- {"command":"read-memory","arguments":{"address":"$sp","length":512},"type":"request","seq":41}
--> {"seq":358,"type":"event","event":"output","body":{"category":"stderr","output":"79-data-read-memory-bytes \"$sp\" 512\n"}}
--> {"seq":359,"type":"event","event":"output","body":{"category":"stderr","output":"-> 79^done,memory=[{begin=\"0x20041fe0\",offset=\"0x00000000\",end=\"0x200421e0\",contents=\"0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000\"}]\n"}}
--> {"seq":360,"type":"response","request_seq":41,"command":"read-memory","success":true,"body":{"startAddress":"0x20041fe0","endAddress":"0x200421e0","bytes":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}}
```

closed 2023-07-16 by meltdown03

meltdown03, 2023-07-16:
Just a followup, the reason using `x/512b $sp` in gdb manually works looks like because it does a bunch of small requests so it doesn't cross the boundary with the first few requests. Maybe that could be an option, but it would probably be slower.