mcu-debug/memview #19: Highlight and copy seems to be off by one

jnz86, 2023-08-09:
**Describe the bug**
Highlighting some number of bytes, then copying does not copy the correct number. The last byte in the highlight will not be on the clipboard. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open an address
2. Click on the first byte
3. Hold shift, and click some bytes away to highlight more than one byte
4. Right click, copy, or `CTRL-C`
5. Paste somewhere

**Expected behavior**
That the number of byte highlighted is copied. Instead the last byte isn't copied. Interesting to me is this works in reverse but instead of the "first" byte missing, it's still the last byte that will be omitted. The address is correctly in the clip board it seems.

**Screenshots**
[image]

For this highlight I have `000000002400bcd0 18 5a 61`

**Environment (please complete the following information):**
 - OS: `Windows 11`
 - Version of extension: `v0.0.23`
 - Debugger Type: `cortex-debug, JLink`
 - Debugger Version: `v1.12.0`

**Additional context**
Right click and `CTRL-C` do the same thing occasionally, but most of the time right click copy does not copy anything from memory view. Depending on what is in another panel, right click copy will take from there not memory view.

Highlighting multiple lines does not work. Copies only the first line, however, when you do that, this error does not happen.

haneefdm, 2023-08-10:
I was able to fix the issue with the last character being dropped.

But, I cannot duplicate the other issues you mentioned.

jnz86, 2023-08-10:
Full disclosure the right click copy is also an issue with cortex-debug's memory viewer. It seems to grab from the wrong pane a lot of the time. So it seems to be something on my VS Code view. Maybe I'll try turning all extensions off except this one and re-try.

haneefdm, 2023-08-10:
The cortex-debug memory viewer is no longer maintained (deprecated). This is the new memory viewer and it works across many debuggers. It may not have the same features but this is the way we want to go.

You are also free to use the [Microsoft memory editor](https://marketplace.visualstudio.com/items?itemName=ms-vscode.hexeditor) but it has its own issues and is not really debugger friendly. Looks like you already tried it as Microsoft integrated it into their Variable/Watch windows. I really did not want to author this extension but I had no choice.

haneefdm, 2023-08-10:
See https://github.com/microsoft/vscode/issues/155597

It already has 45 votes but zero action from Microsoft. You will see my comments in there as well. It is a year old.

jnz86, 2023-08-10:
Yea well, now they have 46! Thanks for the work.

closed 2023-08-10 by jnz86