mcu-debug/memview #20: Data not refreshing

diggit, 2023-10-05:
**Describe the bug**
Memory view is not refreshing data. Not even on after clicking "refresh data" icon. Fresh data are loaded only when a new memory view is created.

**To Reproduce**
1. create memory view of some valid memory address
2. change value from debugger
3. value updates in watch, but not in memory view
4. click update data in memory view, it flashes like it's doing something
5. old data are displayed
6. delete memory view
7. create new memory view of same address
8. displayed data are up to date and match current state

**Expected behavior**
Values update

**Screenshots**
Watch
[image]

Memory view
[image]

GDB command
```
set {uint32_t}0x20000370 = 0x12345678
```
Watch
[image]

Memory view (clicked update data icon)
[image]

Memory view (deleted and created new on same address)
[image]

**Environment:**
 - OS: Arch Linux
 - Version of Memory View (this extension): v0.2.24
 - Debugger Type: cortex-debug
 - Debugger Version: v1.12.1

**Additional context**
J-link, Cortex M33 (STM32L552), non secure

haneefdm, 2023-10-05:
If you change a value while the debugger is stopped, we don't get a notification of that. Or, if it changes some other way. We only refresh when the debugger halts because of a breakpoint/refresh. Sorry.

This is why there is a Refresh button [image] So, are you saying that the Refresh button is not working?

diggit, 2023-10-05:
> If you change a value while the debugger is stopped, we don't get a notification of that.
Watch values get refreshed even on every GDB command. That is not possible for memory view?

> This is why there is a Refresh button image So, are you saying that the Refresh button is not working?
Yes, displayed data remain the same after clicking that refresh icon. Resume/pause/stepping triggers memory view data refresh correctly.

haneefdm, 2023-10-05:
>> Watch values get refreshed even on every GDB command. That is not possible for memory view?

Nope. MS controls the Watch/Variable/Stack windows. No one outside knows what is going on. Not even the debugger/adapter (like Cortex-Debug, cppdbg) knows what is going on. Certain actions in the debug console or variables windows can trigger a refresh...but not to all clients of the debugger.

I will see why the Refresh button not working as intended.

haneefdm, 2023-10-12:
The Refresh button is working just fine for me. I cannot reproduce your problem. We are displaying whatever your debugger is reporting. I am assuming you are using Cortex-Debug. If so,

* Set `"showDevDebugOutput": "raw"` in your launch.json
* Do your normal debugging.
* Make sure your Memory View is exposed and it is displaying the right address/data rows
* Modify the memory value
* Switch to the 'Debug Console',  and clear the contents (Right Click and select `Clear Console`
* Switch to the Memory View and hit the Refresh button [image]

* Switch back to the `Debug Console`. You should see new memory read requests being made. If you don't see that, we have an issue. If you see memory reads, check the values.

If you are using some other debugger, do whatever it takes to do something similar to above.

haneefdm, 2023-11-10:
No feedback...

closed 2023-11-10 by haneefdm