mcu-debug/memview #21: Support for cppvsdbg

mathforlife83, 2023-10-06:
How much work is it to support cppvsdbg, and is it on the backlog?

haneefdm, 2023-10-06:
You can already add your debugger to the list of debuggers. Let us know if `cppvsdbg` works

[image]

haneefdm, 2023-10-06:
This is a setting you can set

mathforlife83, 2023-10-17:
Hi, sorry for the late reply, I have been away on vacation. I have tested adding cppvsdbg in the list, this is the error message that I got:

"The current debugger does provide a memory read API. Cannot add a memory view"

haneefdm, 2023-10-17:
There is your answer. You should file an issue with the cpptools people (Microsoft) who created that debugger. Memory read/write APIs are optional and it should be relatively easy for them to add.

closed 2023-10-17 by haneefdm

murmurlab, 2026-01-01:
[image]

[image]

Okay, I've already added them. But the legacy memory view works, this one doesn't.
and I can't remove the addresses I added, is this normal?

murmurlab, 2026-01-01:
Okay, it worked with cppdbg, I missed that. But the others don't work, so it will be a bit of a disadvantage because I can't use them.

haneefdm, 2026-01-01:
You can debug this in your VSCode a bit. You have to set debug level to >= 1 to see what we are trcking...as you debug. Check the **OUTPUT** tab and select "Mcu-debug Tracker". See if your session is popping up and. Of course, VSCode does not show session ids anywhere so, I cant tell you why we can't talk to your session. It is possible that lldb does not support memory reads (or not implemented by the VSCode adapter) or something else. I can't tell you without duplicating what you have. A sample program (already compiled and if native program, I prefer a Mac)

And, I don't know what you mean by "legacy memory view". We don't have one.

[image]

haneefdm, 2026-01-01:
Unfortunately, there isn't a way to delete the list of addresses stored. It is best to select one and change it to something you want rather than adding a new View. Oversight.

murmurlab, 2026-01-01:
By 'legacy', I mean Microsoft's hex viewer extension. The debugger extension I want to work with is this: vadimcn.vscode-lldb. When I debug level 1 in the logs, it only gave an output like this:
mcu-debug.debug-tracker-vscode-0.0.15: Session 'lldb:code-lldb': Status 'initializing', id = 8f2cdadf-f152-4412-b229-aa88593fcb06
mcu-debug.debug-tracker-vscode-0.0.15: Session 'lldb:code-lldb': event 'capabilities', id = 8f2cdadf-f152-4412-b229-aa88593fcb06
mcu-debug.debug-tracker-vscode-0.0.15: Session 'lldb:code-lldb': Status 'running', id = 8f2cdadf-f152-4412-b229-aa88593fcb06
mcu-debug.debug-tracker-vscode-0.0.15: Session 'lldb:code-lldb': Status 'started', id = 8f2cdadf-f152-4412-b229-aa88593fcb06
mcu-debug.debug-tracker-vscode-0.0.15: Session 'lldb:code-lldb': Status 'stopped', id = 8f2cdadf-f152-4412-b229-aa88593fcb06
mcu-debug.debug-tracker-vscode-0.0.15: Session 'lldb:code-lldb': event 'first-stack-trace', id = 8f2cdadf-f152-4412-b229-aa88593fcb06

Level 2 logs are attached.

[Mcu-debug Tracker.log](https://github.com/user-attachments/files/24401792/Mcu-debug.Tracker.log)

I was just testing it in an empty C program.
thanks

murmurlab, 2026-01-01:
By the way, the reason I use the vadimcn.vscode-lldb plugin is because it has its own disassembly viewer and displays the code in color. In the debug section, there's a "view memory" button next to the registers, so I don't need to copy the address. I can jump to the instruction using the "jump to cursor" option in the right-click menu. etc. It's unfortunate that such simple things aren't available in cppdbg or VS Code UI. Do you have any IDE/platform recommendations for Ubuntu that would cover this?

haneefdm, 2026-01-01:
Well, something that is not ours isn't exactly legacy? That one belongs to Microsoft and hardcoded for them. So, I was confused. Theirs is designed to work mostly with files and is more of a snapshot. Ours updates as the debugger progresses. We don't follow their development anymore. I tried to convince them to support actual debuggers but they refused. Hence  ine appeared.

haneefdm, 2026-01-01:
Whose debugger is "lldb:code-lldb"? That is the session name. What is the actual debugger type and what extension provides that? Never heard of it.

murmurlab, 2026-01-01:
I used the wrong word, "legacy", since it's already installed on my system, I thought it was a VS Code built-in plugin, so I meant the default viewer. I'm translating it, sorry for the eng translation.

> Whose debugger is "lldb:code-lldb"? That is the session name. What is the actual debugger type and what extension provides that? Never heard of it.

i said, extension id: [vadimcn.vscode-lldb](https://marketplace.visualstudio.com/items?itemName=vadimcn.vscode-lldb)

I have no idea how it works. But I think it uses 'lldb-server gdbserver'
`murmur@murmur-IdeaPad-3-15ALC6:~/sources/asm-learn$ ps ax| grep lldb | cat
1817341 ?        Sl     0:05 /home/murmur/.vscode/extensions/vadimcn.vscode-lldb-1.12.1/adapter/codelldb --connect 41375 --auth-token dDvYyupEWmSPVMzhw8UB6Q== --settings {"evaluateForHovers":true,"commandCompletions":true,"sourceLanguages":["asm"]}
1817585 pts/11   S+     0:00 /home/murmur/.vscode/extensions/vadimcn.vscode-lldb-1.12.1/bin/codelldb-launch --connect=127.0.0.1:36305 --clear-screen
1817590 ?        S      0:00 /home/murmur/.vscode/extensions/vadimcn.vscode-lldb-1.12.1/lldb/bin/lldb-server gdbserver --native-regs --setsid --fd=15
1817909 pts/8    S+     0:00 grep --color=auto lldb`

haneefdm, 2026-01-01:
Not every debugger works and not with every backend. They have to support Memory Reads at least which is optional for most debuggers. Microsoft defines the protocol and they HAVE to obey that protocol. That is why it magically works but debugger developers don't advertize what do, do-not support. You have to ask them. Not me, since I didn't write those.

Anyway, you are on your way...