mcu-debug/memview #22: Save memory range as binary file

coran21, 2023-10-28:
I switched to VS Code from Eclipse, and one of the great things there is that you can save a selected memory range as a binary file. Is something like that possible with memview? So far, I could save manually selected memory area as a text file together with additional information, which doesn't help for my use case.

**Describe the solution you'd like**
Add an interface where the user would enter the start and end memory address, select a file, and that memory range would be saved into selected file in binary form.

haneefdm, 2023-10-28:
Need more info:

So, what is a "binary form"? Hex (in text form), or bytes in literally binary that you cannot open in an editor. What is the exact file format?

What is your use case? What does Eclipse do, specifically? Screenshots?

I can consider it if I see a justification. You can have a python/perl/whatever script you could contribute it to this repo as well.

coran21, 2023-10-28:
I think it's called raw binary form, and yes, when you open it in the editor it shows garbage.

My main user case is troubleshooting protocol buffer messages received by MCU and stored in MCU memory. I need to save them as binary files so that I can open them with the protocol buffer parser and find the problem.

The second case is that I work with the camera connected to the MCU. When the image is stored in the MCU memory in jpeg format, it's very convenient just to export memory with that jpeg image and save it directly to the jpeg  file, which I can open on the computer and check the captured image immediately (I understand that this is very rare user case)

I want to say that you are doing a great job and you made VS Code much better for embedded development, thanks for that!

[image]
[image]
[image]

haneefdm, 2023-10-28:
> I want to say that you are doing a great job and you made VS Code much better for embedded development, thanks for that!

Thank you for the kind words. And thanks for providing some context. I can see saving a packet (network, other, etc.)

How about an interface where you can specify any start address (defaulting to the start address of the memory window or cursor-cell position) and a length (not sure what default to use). You can change both and they can even be outside the range of the current memory view. We will TRY to fetch and save.

Both the address and the length can be expressions - that we try to remember for next time. No promises here on the remembering part.

coran21, 2023-10-28:
All of that sounds good. Remembering the last used length would also be nice.

The default start address could be the start address of the selected memory view, and the default length could be 0 (that's in the Eclipse too).