mcu-debug/memview #24: Can't remove entries from drop down address/symbol menu

daguro, 2023-10-31:
**Describe the bug**
The drop down menu with addresses and program symbols on it has 10 entries.  Some of the entries are stale, but the popup menu with three entries (cut, copy, paste) doesn't seem to work.  When I select an item and 'cut', the entry is still there.

**To Reproduce**
Steps to reproduce the behavior:
1. Display 10 different addresses
2. Select an entry on the menu
3. Right click, select Cut and release
4. Entry is not deleted

**Expected behavior**
I expected the entry to be removed.  As it is, the menu has 10 items and won't take any new items via the '+' button.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS:Windows 11
 - Version of Memory View (this extension) 0.0.24
 - Debugger Type cotext-debug
 - Debugger Version  1.12.1

**Additional context**
Awesome tool!  Thanks for doing this.

haneefdm, 2023-10-31:
Did you notice there is an 'x' button to the far right of the toolbar? That does not work? I followed kind of the style used by VSCode elsewhere.

> Awesome tool! Thanks for doing this.

Thanks.

closed 2023-11-10 by haneefdm

nidrissi, 2024-02-24:
@haneefdm I didn't notice it either myself. If I may make a suggestion, I would say that putting the view delete button closer to the add one would be clearer. Anyway, thanks a lot for this extension, it's great!