mcu-debug/memview #25: Convert/Duplicate Default Debugger Memory Views (HexEditor) to a Memory View

haneefdm, 2023-11-10:
**Is your feature request related to a problem? Please describe.**

VSCode provides a mechanism to launch a memory viewer for certain items in the Variables/Watch panels. It users its own HexEditor to provide the visualization. There is no user-level customization provided by VSCode to change which extension/visualizer/editor should be used to display the memory contents. We have been in discussion s with Microsoft for over a year and got many votes but there is no interest from them in providing such a customization

**Describe the solution you'd like**

As a stopgap, could we detect a new HexEditor being launched and convert that into a view in the memory panel?

**Describe alternatives you've considered**
Tried to get the HexEditor to be more debugger-friendly, so this extension is not even needed. 2+ years and here we are. It is focused on editing files

**Additional context**
The following issues we have filed with VSCode. We made near-zero progress

https://github.com/microsoft/vscode/issues/155597
https://github.com/microsoft/vscode/issues/197287
https://github.com/microsoft/vscode/pull/197458

haneefdm, 2023-11-10:
I implemented the following settings and what you see below are the defaults. You probably installed our extension because you are more interested in seeing memory views done better. It should be released shortly

[image]

Please vote to give us feedback. Like, hate, useless, distracting, improvements, etc. Use emojis in this box but please also leave feedback if you have suggestions focused on this topic

muliu, 2023-12-18:
[image]
How to attach the memory viewer to Debugger ?

haneefdm, 2023-12-18:
Not sure what debugger you are using.

See https://github.com/mcu-debug/memview/issues/21