mcu-debug/memview #29: Allow to limit the size of a memory region

andreasWallnerIFX, 2024-07-11:
In cases where I display memory with limited accessibility (e.g. register area, limited via MPU, etc.) it can be quite annoying to always get error message during scrolling for memory that could not be read.

## Describe the solution you'd like

It would be cool if in addition to providing a start address we could also be able to set an (optional) size / end address for the display.

## Describe alternatives you've considered

I don't see a real alternative other then ignoring the error message ;-)

haneefdm, 2024-07-11:
Yes, makes sense. But the issues go in both directions. You can also scroll to before the start address (maybe accidentally). Yes, I will think of a min/max boundaries for scrolling and viewing. I have another suggestion.

It has been a while since I have looked at this but your previous session's status is stored along with your workspace's private user data.

What if the(dis)allowed regions are stored along with your project or workspace? It could be a user/workspace setting, or a JSON file (perhaps alongside launch.json) that can be maintained by the user. If that file is checked into version control, then the entire team could benefit from it.

andreasWallnerIFX, 2024-07-17:
Being able to store them might be nice, but I'm not sure it's a good fit for my use-cases: if a memory region is locked via some MPU-like peripheral, accessbillity will change depending on configuration. That's why I was thinking of being able to provide this as a parameter.

Your solution makes total sense for other use-cases: hardware registers where the read has a side-effect (which one could actually parse from the SVD if it's complete & correct), general memory boundaries,