mcu-debug/memview #36: Modifying settings.json to add debuggers

JacobYiu, 2025-01-27:
Hi, I wanted to use MemoryView with the debugger I am currently using, [LLDB](https://github.com/vadimcn/codelldb). I saw that in the issues section that I had to add something in the settings.json in order for MemoryView to work with my debugger. But I am not sure what I should be adding here

'''
{
    "cmake.showOptionsMovedNotification": false,
    "hexeditor.columnWidth": 16,
    "hexeditor.showDecodedText": true,
    "hexeditor.defaultEndianness": "little",
    "hexeditor.inspectorType": "aside",
    "memory-view.trackDebuggers": [
        
    ]
}
'''

haneefdm, 2025-01-27:
lldb (like gdb) does not support Debug Adapter Protocol and is not a debug adapter. Which (visual) debugger are you actually using? What does your launch.json say for `"type"`?

JacobYiu, 2025-01-27:
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [

        {
            "type": "lldb",
            "request": "launch",
            "name": "Debug",
            "program": "${workspaceFolder}/hello",
            "args": [],
            "cwd": "${workspaceFolder}"
        }
    ]
}
```

It says lldb as the type. 

I thought you had managed to make it work using lldb when someone raised it though
[LLDB Help](https://github.com/mcu-debug/memview/issues/15)

Unless I may be mistaken and misunderstood

haneefdm, 2025-01-27:
No, we made it work with a debugger extension front end that uses `lldb`. That was Microsofts `cppdbg`.  So, it depends on the extension (front end). Debugger extensions advertise the type(s) of debuggers they support. `cppdbg` supports 3-4 but we don't care, the main type is `cppdbg`

Try the following in your settings and re-start VSCode

```
    "memory-view.trackDebuggers": [
        "lldb"
    ]
```

From #15,  I saw the extension link and found the following debugger this extension it provides
https://github.com/vadimcn/codelldb/blob/3b196affdefffbccafe409d0c1d07cdf1d08a0f1/package.json#L674

You should not confuse the extension debugger type with the actual debugger. In this case, they are named the same.

Are you using a different debugger extension?

JacobYiu, 2025-01-27:
Oh I see. I believe I may have confused the extension debugger type with the actual debugger. I am running on VSCode and I am using the CodeLLDB extension which follows the debug adapter protocol. Once I added in lldb it is working perfectly and I am able to use the MemoryView extension. Thank you very much!