mcu-debug/memview #37: memory view does not refresh firmware flash content

TwoBrushes, 2025-02-05:
**Describe the bug**
1、memory view does not refresh firmware flash content
2、sometimes memory view displays ~~

**To Reproduce**
Steps to reproduce the behavior:
when i use IAP to uprade firmware, but the memory view always disply init value,  can't refresh flash content

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Version of Memory View (this extension) [e.g. 0.2.3]
 - Debugger Type [e.g. cortex-debug, cppdbg, etc.]
 - Debugger Version [e.g. 1.12.1]

**Additional context**
Add any other context about the problem here.

haneefdm, 2025-02-05:
Why didn't you fill out the details of your environment. Please repoen after providing details.

Did you try the debugger to see if it is returning the same results all the time?

TwoBrushes, 2025-02-05:
OS: windows 11 23H2
Version of Memory View (this extension) : v0.0.25
Debugger Type [e.g. cortex-debug, cppdbg, etc.] : cortex-debug
Debugger Version : v1.21.1

Did you try the debugger to see if it is returning the same results all the time?

i set "showDevDebugOutput": "raw" in launch.json,  and when refresh the memory, in debug consle,  the displayed memory content is outdated, It seems to get cached memory content.
the same project in MDK debug,  it's ok!

haneefdm, 2025-02-05:
It depends on the debugger. If your linker script marked the flash area as **read-only**, then gdb will not go to the device to fetch any data from that region. It gets it from the executable once. Nothing I can do.

Google it and see how to can force gdb to read/update from flash (or any read-only memory). Also, post here if you found a workaround.

Also, if you mark flash as read-write, then your brreakpoints may not work because gdb will attempt to use SW breakpoints. So, be careful.

TwoBrushes, 2025-02-05:
thanks for replay!
I want to know if this issue is caused by cortex-debug or jlink gdb server? If it is caused by cortex-debug, is there any way to fix it?

haneefdm, 2025-02-05:
None of the above. **_It is GDB_**. You can issue gdb commands in the Debug Console and see what you get.

Cortex-Debug never talks to JLink SW. 100% of its requests go directly to GDB. And, memory-view never talks to gdb either. It only talks to Cortex-Debug. Not JLink's fault either because if GDB does not ask JLink gdb-server to read the memory, then it won't.

Memory-view -> cortex-debug -> gdb -> JLink GDB server -> .... -> Your chip